Rikkert-RS/VenusOS-TemperatureService #3: Phantom of the temperature probe

Fraser86, 2023-03-25:
I recently got this running with a ds18b20 sensor. 
Fantastic and simple when you know how. 

I went further by removing my probe and adding three sensor together with a 4.7k resistor between the plus and the signal wire. Daisychained.
It works fine aswell. 

However the first cae which I named Kitchen is still present. With a temperature of -1c brrrr. 

I dont see why its there and how to remove. It from my device list
Anyone know whats going on

Fraser86, 2023-03-25:
(Cae) should be probe

Fraser86, 2023-03-25:
[image]

Fraser86, 2023-03-25:
[image]

Fraser86, 2023-03-25:
[image]

SirUli, 2023-03-26:
I guess you have restarted the OS after you have changed the sensors, right? Theoretically the sensors which are no longer available should be possible to be deleted from the overview menu page.

Fraser86, 2023-03-26:
Yes. First thing i did was reset. Hoping it would clear. But stays. I dont see anywhere to delete it in overview menu. Also not in setup within that device

Fraser86, 2023-03-26:
Was thinking of solering that sensor into the daisy chain to see it is ID is recognised and temp is displayed. 

Question is still how to choose which temps are displayed and not.

Fraser86, 2023-03-26:
I reset my system several time. But it remains. 
In the device list there is sometimes the option to remove un-used devices but this is not present. I guess its because this probe is still seeing -1c. 

If i tap on the device (kitchen)
I see the setup and the device pull down menus. 
Setup gives me option to choose battery, fridge or generic. 
Device shows the attached image. 
No remove option visable to me 

[image]

Rikkert-RS, 2023-03-27:
Hello Fraser 86

I was able to reproduce your case today and will look for a solution tomorrow.

Best regards, Rikkert-RS

Rikkert-RS, 2023-04-24:
Hello Fraser86,

I just found time to solve the problem.

There was an error reading the temperature value. This is fixed in version 0.5
https://github.com/Rikkert-RS/VenusOS-TemperatureService/releases/tag/latest

If you remove a sensor, it will show as Disconnected after about 2 to 5 minutes. If you then do a reboot, the sensor is no longer there until you reconnect it

Rikkert-RS, 2023-04-24:
**Attention** completely forgotten. 
Please uninstall version v0.4 in the setup first before installing version 0.5 
because the GPIO PIN has been changed from **4 to 26**

Rikkert-RS, 2023-04-26:
If you still have problems, just write again

Fraser86, 2023-05-02:
I am having problems to get this to work. 
I have moved over to a test pi. I have this so i dont end up without my solar monitor. 
I playing with the relays and temps. 
But i have moved to gpio26 and 5v and ground. But the temperatures are not showing. 
Firmware was 3.0. Now running 2.94 to see if that work. 
Rebooted many times. But temperature reading not showing

Fraser86, 2023-05-02:
[image]

Fraser86, 2023-05-02:
[image]
Only cpu showing

Rikkert-RS, 2023-05-02:
OK that's not so good, but I think I can help you but for that I need some more information.
First test Pi is a good solution ;)

1. Have you installed RpiGpioSetup? 
If yes please uninstall and check again. The reason for this is that it is very difficult to always find the right free GPIO that does     not collide with other things.
If it works then you just have to create a custom GPIO list for RpiGpioSetup that's not difficult

3. What kind of sensor have you now installed?

4. if it is a DS18B20 how it is connected?
Don't forget the pull-up resistor
Example image:
[image]

5. . Can you access your Pi via SSH to read a few some log files?

Rikkert-RS, 2023-05-02:
Hello, I have found something after all. 
Because you have auto update active, most likely the pin was not changed from 4 to 26 in the config. Here is a solution without you have to do something via SSH. 
Go to the Packagemanager and change the tag from "latest" to "v0.4" -> downlaod -> install. 
Then unistall the v.04
After that change the tag again to latest -> download -> install.

then i hope pin26 is running. If not i need the logs

[image]

Rikkert-RS, 2023-05-21:
no answer anymore problem is probably solved

closed 2023-05-21 by Rikkert-RS