Rikkert-RS/VenusOS-TemperatureService #5: Can't find 1-wire sensor

Mange131, 2023-05-12:
The sensor worked when i install it then after couple of minutes it says, it's not connected. Can i check in someway if the sensor is there?

Rikkert-RS, 2023-05-21:
Hello Mange131,

can you please tell or describe how you connected the sensor?

mfg Rikkert

Mange131, 2023-05-21:
i use this sensor with pluggable terminal adapter with build in  4.7 pullup
resistor.

https://www.amazon.se/gp/aw/d/B094FKQ9BS?psc=1&ref=ppx_pop_mob_b_asin_image

 Data to gpio 26, 3,3v and earth.
 I have tested the pullup resistor after, and it's ok

Venus os 3.00-39.

I have testet two times, from scratch. Because i have probleme with the rpi
can't find my hdmi screen.  second time it work for 2-3 hour till i install
rpigpiosetup.

I have rpi in my boat. Can test it tomorrow, same time as now?

Mvh/ Magnus Eriksson tel. 070 224 77 98

Rikkert-RS, 2023-05-21:
Hello,

ok i think the issue is with rpigpiosetup. you have to modify the gpio list and exclude the gpio 26.

The reason why I changed the pin is that I installed a can hat for a canbus battery and that unfortunately collides. Pin 4 could also cause problems for other people and other raspberry shields.

If it works then you just have to create a custom GPIO list for RpiGpioSetup that's not difficult.

Link->[https://github.com/kwindrem/RpiGpioSetup]

[image]

here a view from the RpiGpioSetup gpio_list file: were you can comment out the Gpio26:
do that with ssh under /data/RpiGpioSetup/FileSets/gpio_list

[image]
[image]

Maybe in the future i can make the pin configurable, or comment out at installation if RpiGpioSetup ist installed

Mange131, 2023-05-21:
Ok i test that tomorrow

Thanks

Mvh/ Magnus

Rikkert-RS, 2023-06-12:
@Mange131 
Did you already have time to test the problem is solved

Mange131, 2023-06-14:
only worked briefly again.  maximum one hour.  tried to uninstall and install temperatureService again but then the sensor did not come up.  have installed guimods, rpidisplaysetup. Shutdown manager

Rikkert-RS, 2023-06-14:
Hi,

Hmm that is not so good. 
How long is your Sensor cable?

You can try an other Port with this method: 
https://github.com/Rikkert-RS/VenusOS-TemperatureService/issues/6#issuecomment-1587847440

otherwise we must check out the logs

Mange131, 2023-06-14:
ok, going to the boat later today.  I have 2 sensors of 1 and 3 meters to one pullup 4.7k and then from the pullup I have a one cable of 5 meters to the rpi. slightly thicker cable.  Do i need uninstall something?

Rikkert-RS, 2023-06-14:
Ok 

if you want to switch the Port you don‘t need to uninstall only connect with ssh and change the config.txt file an restart the rpi. 

But I would test the sensors with a shorter cable length beforehand. The voltage drop at 3.3v on the cable length can be considerable.
If the cable is also not shielded, this can also lead to interference if other cables are nearby.

Rikkert-RS, 2023-06-14:
Some info for 1 Wire bus wiring

http://pvlng.com/1-Wire_Hub
https://www.analog.com/en/technical-articles/guidelines-for-reliable-long-line-1wire-networks.html
https://www.msxfaq.de/sonst/bastelbude/raspi/raspi-1wire.htm

Good choice is to use Cat 5 or 6 Network cable

Rikkert-RS, 2023-07-02:
Interesting, is the sensor running with the temperature values? 
Which Venus OS version are you using?

Mange131, 2023-07-02:
What do you mean with temperature values? sry English is not my language 😀
I'm using 3.0

Just in the process of running a cat6 cable for the sensors with longer distances

Rikkert-RS, 2023-07-02:
No problem, English is not my native language either.

I meant if only the metadata to the sensor is missing or also the sensor value, so the temperature value? 
Possibly it is the version 3. At the moment I have still no device runs on V3.0 to test.

For 
> Just in the process of running a cat6 cable for the sensors with longer distances

You can also reduce the pullup resistor from 4.7kOhms to something less, perhaps 3.3kOhms or 2.2kOhms, since the RPI pin voltage is 3.3V.

Mange131, 2023-07-02:
now both sensors work with the longer cat6 cable.  we'll see how long 😀 and the id number comes up as well
[image]

Mange131, 2023-07-02:
[image]

Rikkert-RS, 2023-07-02:
ok, ;) inform me if it's working properly the next days

Mange131, 2023-07-02:
Ok i do that.
 thanks a lot for your help

Mange131, 2023-07-12:
only one works for me now.  I will check the cable connection.  think it's all to do with sensor instability.  if you search online about rpi and 1-wire.  so you get quite a lot of hits on those who have problems.  one solution to having more sensors is to maybe be able to have more pins available.

We can probably close this thread now.  thanks so much for all the help

closed 2023-07-12 by Rikkert-RS