Rikkert-RS/VenusOS-TemperatureService #6: Seems to have issues with VenusOS 3.00

twaldenmeier, 2023-06-12:
Hey,
Any idea if this is working with VEnusOS 3.0?

On my VRM Console it does switch beteween the following two screens within a second from this:

[image]

to this:

[image]

Is this a v3.00 issue or maybe a setting issue?

Rikkert-RS, 2023-06-12:
Hello,

I haven't had time to update my Venus OS to version 3 yet. So I can't test it or reproduce it.
Question: On which device do you have it installed?

best Regards Rikkert-RS

twaldenmeier, 2023-06-12:
Thanks for your quick reply!

I'm using a RPi 3B+

Rikkert-RS, 2023-06-12:
Yes I have now also seen stupid question from me.
Another question: How are the sensors connected? Which ones are installed?

Oh yes that often gives problems when Setuphelper packages are installed that interfere with each other.

Which are installed?

twaldenmeier, 2023-06-12:
My 1 Wire temp sensor is connected to this board and from the board it connects to the RPi: 
[image]

[image]

Rikkert-RS, 2023-06-12:
Ok connection looks ok for me. Pull up resistor is on the extra little Board right?
Sensor is DS18B20?

It seems that the Service is restarting again and again.

Is SSH access possible for you? Then we can get more out through the logfile

twaldenmeier, 2023-06-12:
Yes, that's the sensor: https://amzn.to/3P2SsNb

This is what I've installed from setuphelper:
[image]

This is your package:
[image]

Rikkert-RS, 2023-06-12:
Oh i think is see the problem.

The problem at the moment is that the GPIO pin 26 collides with the "RpiGpioSetup"
You can use this both but only with mods in "RpiGpioSetup"
For this have a look at: https://github.com/Rikkert-RS/VenusOS-TemperatureService/issues/5#issuecomment-1556277827

For testing deinstall "RpiGpioSetup" and reboot you device

twaldenmeier, 2023-06-12:
That make sense! I thought 26 sounded somehow familiar. But since the Relay Board needs to use Pin 26 - do I have the chance to change your package data pin to for example Pin4?

Rikkert-RS, 2023-06-12:
Oh yes we can test something I'll build a pack with pin 4 and I'll create a tag for it. need some time for it 30min or so.

Alternatively it would also work if you connect via SSH and change the pin in the Config.txt

Config file is also possible but then you must always have SSH access.

Rikkert-RS, 2023-06-12:
Looks good but i do not quite understand what has been changed now?

I think the best convenience is to build it into the Venus menu. I am working on it but unfortunately it is a bit more time consuming

twaldenmeier, 2023-06-12:
I did not change anything else - at least I'm not aware of ;-) . I just tried to re-install the package and change to Pin 4 again - even now the custom names are lost ;-) 

[image]

But for now this helps a lot and I understand the dependency ... Will you be able to adjust this to make it a little more customizable?

Rikkert-RS, 2023-06-12:
yes as I wrote above it is possible

> I think the best convenience is to build it into the Venus menu. I am working on it but unfortunately it is a bit more time consuming

is "RpiTemperature" still installed?

twaldenmeier, 2023-06-12:
No, the "RpiTemperature" is not installed anymore.

Before restarting (I did change the names):
[image]

After Restart, looks good:
[image]

Rikkert-RS, 2023-06-12:
OK looks good for the moment,

i try to update my venus OS the next days and make the pin configurable in V0.6
You can see this in the setup helper when a new version is available

have a nice evening

Rikkert-RS

twaldenmeier, 2023-06-12:
Thank you so much! :-)

closed 2023-06-12 by Rikkert-RS