Rikkert-RS/VenusOS-TemperatureService #7: Two or more DS18B20 temp sensors

twaldenmeier, 2023-06-16:
Can I add two or more DS18B20 temp sensors and they will show up on the console?

Rikkert-RS, 2023-06-16:
Hello, yes it is a Bus system

look at this Post: 
https://github.com/Rikkert-RS/VenusOS-TemperatureService/issues/5#issuecomment-1591694233

twaldenmeier, 2023-06-16:
Thank you! I will give it a try!

twaldenmeier, 2023-06-18:
I have connected a second sensor but I am not sure if I configured the config.txt correct since it does not show up on the console. The custom name disappeared and this is what it looks like with even two sensors connected

[image]

This is the change I made ... second sensor isconnected to Data Pin 4 (GPIO).
[image]

Any idea what I did wrong?

Rikkert-RS, 2023-06-18:
Yes this is wrong 1Wire is a bus system you need only 1 Pin for that.
Do you have read my links posted before? Delete the line with pin 26 in the config.txt

You have to connect the second or more Sensors in parallel to the first one. Data pin is for all Sensors GPIO 4 in your case.

[image]

Rikkert-RS, 2023-07-02:
There are many possibilities, wire yourself or various boards

closed 2023-07-12 by Rikkert-RS