Rikkert-RS/VenusOS-TemperatureService #8: Modbus Error for CPU Sensor

SvenTej, 2023-08-04:
First of all: Cool stuff, thanks for it.

I connected 3 18B20 sensors and up to now it works like a charm.

The only issue I have:
I read the temps from my Siemens LOGO! 8.3 SPS to use it as inputs for a fan control in my control cabinet.
This reading is done via MODBUS TCP.

It works for the 1 wire sensors but not for the CPU.

The CPU Temp states to be device 100 (system?) and if I use 100 as UID for the MODBUS the Venus OS shows an Error that it cannot find the function.

Is this a known issue?

Greets from Stuttgart, Germany

SvenTej, 2023-08-06:
After I used the data now and tried to implement some error handling for broken sensors in the PLC I detected some additional issues:

If I disconnect a sensor
After about 5 seconds, the temp in the devicelist shows "OK" (not the temp any more), the value in com.victronenerg.temperature, dbus-obj-path \Status (Modbus ID 3305) shows still 0, whereas the temperature (dbus-obj-path Temperature, Modbus 3304) shows already 0.
After about 90 seconds after that, the Status chenges to 1 and display shows "unconnected".

If I then reconnect it in some cases I saw status going to 0 some seconds before tempreature has the correct value.

Is this by design?

S.

SvenTej, 2023-08-10:
What I now also found:
After each restart, the Modbus Device IDs change.
So the device ID of a sensor is not preserved after a restart

Rikkert-RS, 2023-11-29:
Hello,

I don't use the Modbus service yet. I have to check what can be done there. Maybe it's an ID conflict. I have to build a test setup first

best regards Rikkert-RS

closed 2025-01-11 by Rikkert-RS

SvenTej, 2025-05-12:
Hey, 
What was the outcome?

Can I preserve the ModBus ID even if I add sensors?