Rikkert-RS/VenusOS-TemperatureService #11: The DS18B20 sensors disconnects after some time.

mcfrojd, 2023-09-26:
I am running a Raspberry Pi 3b+ with the latest release candidate v3.20~4.
[image]

( I have the same problem on my production system, a pi 4 with v3.01 )
VenusOS-TemperatureService version 0.5
Using two DS18B20 temperature sensors with a 4.7kohm resistor.

I have no trubbel setting them up and using them for a couple of days.
But after some time the two DS18B20 temperature sensors will end up disconnected.

And this is the only thing i find in the logs at /data/log/VenusOS-TemperatureService/current

```
@40000000650cb9a5242d8efc INFO:root:1Wire Sensor found with no Service -> Create:
@40000000650cb9a5281d572c INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.Wire_id03
@40000000650cb9a52944cfcc INFO:root:Add setting /Settings/Temperature/3de104572da1/TemperatureType [0, 0, 3]
@40000000650cb9a52958d8b4 INFO:root:Add setting /Settings/Temperature/3de104572da1/CustomName ['', 0, 0]
@40000000650cb9a539272c14 INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 2
@40000000650cb9a539324fa4 INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName CPU
@40000000650cb9a5393ce694 INFO:root:Retreived setting /Settings/Temperature/3de1045750fa/TemperatureType /TemperatureType 2
@40000000650cb9a539510ad4 INFO:root:Retreived setting /Settings/Temperature/3de1045750fa/CustomName /CustomName Probe1
@40000000650cb9a539512a14 INFO:root:Retreived setting /Settings/Temperature/3de104572da1/TemperatureType /TemperatureType 2
@40000000650cb9a5396788a4 INFO:root:Retreived setting /Settings/Temperature/3de104572da1/CustomName /CustomName Probe2
@40000000650cb9a5397d6e1c INFO:root:Created Service 1Wire ID: 3 Settings ID:3
@40000000650cb9c004b3ae94 INFO:root:W1-temp:28-3de1045750fa temperature interface connected
@40000000650cb9c0053d29e4 INFO:root:W1-temp:28-3de104572da1 temperature interface connected
@40000000650cb9c00580e5bc INFO:root:Connected to dbus, and switching over to GLib.MainLoop() (= event based)
@400000006512b44d223a90cc INFO:root:W1-temp:28-3de1045750fa temperature interface disconnected
@400000006512b4551efcea2c INFO:root:W1-temp:28-3de104572da1 temperature interface disconnected
```

[image]

The CPU temperature keeps connected.

Is there any way to find out why they get disconnected?
A reboot wount make them come back to life, instead they disappear.

Running setup with option "i" wount make them come back to life.
But if i disconnect the 3.3v pin and connect it again they will come back.

This is in the log after a reboot, running setup with the option "i" and the disconnect and reconnect the 3.3v pin
```
@400000006512b44d223a90cc INFO:root:W1-temp:28-3de1045750fa temperature interface disconnected
@400000006512b4551efcea2c INFO:root:W1-temp:28-3de104572da1 temperature interface disconnected
@40000000651300573421aa3c *** CCGX booted (0) ***
@400000006513005b1633960c INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
@400000006513005b1715725c INFO:root:Loglevel set to INFO
@400000006513005b198e4a2c INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.RPi_cpu_id01
@400000006513005b198e713c INFO:root:Add setting /Settings/Temperature/1/TemperatureType [0, 0, 3]
@400000006513005b198e84c4 INFO:root:Add setting /Settings/Temperature/1/CustomName ['', 0, 0]
@400000006513005b22437624 Traceback (most recent call last):
@400000006513005b22439564   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 403, in <module>
@400000006513005b2243a8ec     initSettings(newSettings)
@400000006513005b2243b0bc   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 263, in initSettings
@400000006513005b2243c444     settings = SettingsDevice(
@400000006513005b2243d7cc   File "/opt/victronenergy/dbus-modem/settingsdevice.py", line 51, in __init__
@400000006513005b2243f70c     raise Exception("The settings service com.victronenergy.settings does not exist!")
@400000006513005b2245cfb4 Exception: The settings service com.victronenergy.settings does not exist!
@400000006513005d19863f94 INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
@400000006513005d1991305c INFO:root:Loglevel set to INFO
@400000006513005d1a1062dc INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.RPi_cpu_id01
@400000006513005d1a46b134 INFO:root:Add setting /Settings/Temperature/1/TemperatureType [0, 0, 3]
@400000006513005d1a51d8ac INFO:root:Add setting /Settings/Temperature/1/CustomName ['', 0, 0]
@400000006513005e30d895d4 INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 2
@400000006513005e30ef65ac INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName CPU
@400000006513005e310533b4 INFO:root:cpu temperature interface connected
@400000006513005e315d3adc INFO:root:Connected to dbus, and switching over to GLib.MainLoop() (= event based)
@40000000651300c503555ee4 VenusOS-TemperatureService/setup: --- starting setup script v0.5
@40000000651300cb2ac57c5c VenusOS-TemperatureService/setup: ++ Installing VenusOS Temperature Service
@40000000651300cb2d0fd1ec VenusOS-TemperatureService/setup: restarting VenusOS-TemperatureService service
@40000000651300cb3159915c VenusOS-TemperatureService/setup: restarting VenusOS-TemperatureService logger
@40000000651300cb391405bc VenusOS-TemperatureService/setup: completed
@40000000651300cc2a8e5344 INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
@40000000651300cc2a94891c INFO:root:Loglevel set to INFO
@40000000651300cc2bb14b8c INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.RPi_cpu_id01
@40000000651300cc2be6172c INFO:root:Add setting /Settings/Temperature/1/TemperatureType [0, 0, 3]
@40000000651300cc2bef4e8c INFO:root:Add setting /Settings/Temperature/1/CustomName ['', 0, 0]
@40000000651300cc2f869294 INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 2
@40000000651300cc2f96bf34 INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName CPU
@40000000651300cc2fa7d24c INFO:root:cpu temperature interface connected
@40000000651300cc2ffde574 INFO:root:Connected to dbus, and switching over to GLib.MainLoop() (= event based)
@40000000651300fe310431e4 INFO:root:1Wire Sensor found with no Service -> Create:
@40000000651300fe31b7c2a4 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.Wire_id02
@40000000651300fe31fa7cac INFO:root:Add setting /Settings/Temperature/3de1045750fa/TemperatureType [0, 0, 3]
@40000000651300fe323eb584 INFO:root:Add setting /Settings/Temperature/3de1045750fa/CustomName ['', 0, 0]
@40000000651300fe35fa3eb4 INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 2
@40000000651300fe35fe8474 INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName CPU
@40000000651300fe3608c95c INFO:root:Retreived setting /Settings/Temperature/3de1045750fa/TemperatureType /TemperatureType 2
@40000000651300fe361bad4c INFO:root:Retreived setting /Settings/Temperature/3de1045750fa/CustomName /CustomName Probe1
@40000000651300fe364691c4 INFO:root:Created Service 1Wire ID: 2 Settings ID:2
@40000000651300ff311817a4 INFO:root:1Wire Sensor found with no Service -> Create:
@40000000651300ff31c0908c INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.Wire_id03
@40000000651300ff323562cc INFO:root:Add setting /Settings/Temperature/3de104572da1/TemperatureType [0, 0, 3]
@40000000651300ff323fab9c INFO:root:Add setting /Settings/Temperature/3de104572da1/CustomName ['', 0, 0]
@40000000651300ff37638ae4 INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 2
@40000000651300ff376c492c INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName CPU
@40000000651300ff377526b4 INFO:root:Retreived setting /Settings/Temperature/3de1045750fa/TemperatureType /TemperatureType 2
@40000000651300ff377d873c INFO:root:Retreived setting /Settings/Temperature/3de1045750fa/CustomName /CustomName Probe1
@40000000651300ff3785f764 INFO:root:Retreived setting /Settings/Temperature/3de104572da1/TemperatureType /TemperatureType 2
@40000000651300ff37d27d3c INFO:root:Retreived setting /Settings/Temperature/3de104572da1/CustomName /CustomName Probe2
@40000000651300ff381d16e4 INFO:root:Created Service 1Wire ID: 3 Settings ID:3
@40000000651301002e740184 INFO:root:W1-temp:28-3de1045750fa temperature interface connected
@40000000651301002ecb261c INFO:root:W1-temp:28-3de104572da1 temperature interface connected
```

Rikkert-RS, 2023-11-29:
Hello,

It seems like the 1Wire bus is stuck.
When the 1Wire no longer reports the sensor, I set the status to “Disconnected”.
Can you describe your setup to me in more detail? So cable length, cable type etc.

best regards Rikkert-RS

Sarah-1331, 2024-01-06:
I have just had the same installed the first one all fine 
Added two more and the extra 2 have disappeared 

wire 1 5mtr long 4.3k resistor cut at 4m and added 2 more 
Wire 2 1m
Wire 3 1m
So all are 1m from the “junction” but only the first shows worked for like a day

Rikkert-RS, 2024-01-06:
Hello @mcfrojd Is the problem still existent?

Sarah-1331, 2024-01-09:
Hi Rikkert I do have an issue related to this 

I have one sensor absolutely fine 5mrt long 4.7k resistor 
I have added two more 1mtr sensors at 4m down the 5m now have three at 1m at the end of a 4m extension

the two new ones have disconnected any advise wires are good to me 

thanks 
Paul

Rikkert-RS, 2024-01-09:
Hello,

This is most likely not due to the driver itself but rather the cable length/type or the wiring of the sensors.
Do you have a sketch of your wiring and the type of sensors for me?

Sarah-1331, 2024-01-10:
@Rikkert-RS  they just disconnected again but weirdly it’s just the two new ones any advise ?

Rikkert-RS, 2024-01-10:
Hello i can only help if you send a sketch of your wiring / lenght / calbletyp and the type of sensors

Sarah-1331, 2024-01-10:
Sorry it’s not good of a drawing am a builder not an artist 

temp 1 is the original wire that was 5m but now has a “junction in it”
I don’t have the wire thicknesses but temp 2 and 3 are slightly thinner 

sensors I don’t have a brand only ds18b20 they were off eBay 

[image]

Rikkert-RS, 2024-01-10:
Twisted cables are mandatory for 1 Wire!

Sarah-1331, 2024-01-10:
The standard cable it came with I obviously don’t know the wire size of that. I think it’s the thinner wire of the two additional ones as it’s thinner and the original one still works. but it’s not run with any power cables

Rikkert-RS, 2024-01-10:
Some 1Wire Guidelines:
https://www.analog.com/en/technical-articles/guidelines-for-reliable-long-line-1wire-networks.html

Connection example with 5V for Raspi because all Pins only 5V tolerant!
https://www.msxfaq.de/sonst/bastelbude/raspi/raspi-1wire.htm

Rikkert-RS, 2024-01-10:
> The standard cable it came with I obviously don’t know the wire size of that. I think it’s the thinner wire of the two additional ones as it’s thinner and the original one still works. but it’s not run with any power cables

I think the problem is that the extension cable is not twisted or the sensor power supply at the end is bad because of the voltage drop (thin wire gauge).

Rikkert-RS, 2024-01-10:
But you don't have a board like that between, do you?

[image]

Sarah-1331, 2024-01-10:
No I don’t have that board I soldered the wires together and have this on the pi 
[image]

Rikkert-RS, 2024-01-10:
ok, that's good. Your Setup is ok, but i think your extension cable is not good enough

If the driver says that the sensor is not connected but others are still there then it is due to the signal quality and this depends on the cable and wiring. Can you swap the 4m cable for a better one? Network cable Cat5 or better.

Sarah-1331, 2024-01-10:
So that’s what I’ll do I’ll order one now 
I have just put VCC TO 5v and signal -4.7k to 3.3 while I await but I fear it’s the new sensors as the cables are thin

morphias2004, 2024-01-21:
I have the same issue.

I mine is connected directly to the Pi without any extensions on the cable.

The disconnect is random. Sometimes it will be hours, other times it will be days.

The main issue is that you have to shut down and remove power from the Pi to get the sensor back.

Sarah-1331, 2024-01-21:
I have the same I have 5 connected directly at the pi now and they are doing the same except one I have purchased from multiple sellers to try eliminating “cheap” ones

Rikkert-RS, 2024-01-21:
Hello everyone,

So I can't reproduce the error on my side at the moment.
Therefore I need more information from your side whether it is my driver or the 1Wire driver / overlay file or a wiring problem.
For both (@Sarah-1331 and @morphias2004) please post your log files.

To get the log file you need to login with SSH to your Venus OS
 inside the ssh session run this command: 
 `tail -n 200 /data/log/VenusOS-TemperatureService/current | tai64nlocal`

@morphias2004 

1. Which Raspberry Pi Version?
2. Which Version of Venus OS?
3. Which Version of VenusOS-TemperatureService?
4. Which Pin you using?
5. Which cable / wiring?
6. How many sensors and how connected (wired)?
7. Do you have other drivers that also use pins?

@Sarah-1331 

Have you already switched to the other cable?

If I can make it today, I will test a 50 meter network cable today. I will connect 2 or 3 sensors to the end of the cable for testing.

Sarah-1331, 2024-01-21:
I’ll power off and on then wait for it and log it today 

I decided to put the sensors close to the pi and eliminate the “extension” I purchased a few more to eliminate “cheap” sensors 

I currently have 7 sensors
5 x 1m
3 x 5m

I have all the ground wires split up 
All data to gpio 18 with a 4.7k resistor to the same pin at the pi 
I have 2x 1m sensor  and 1x 5m sensor to 3.3v 
I have 3x 1m sensors and 2x 5m sensors to 5v 

will log what goes down first 

one thing I have noticed is in VRM once it disconnects the device shows as “last seen” just minutes ago the entire time but no temp is recorded 

if i reboot not power down it then shows as last seen xx hours ago 

if i power cycle everything reappears

Rikkert-RS, 2024-01-21:
@Sarah-1331 

OK first of all nothing unusual in your log file the error at the start with the "settings" is normal there must be a waiting loop because the Venus OS settings service is not yet ready after a reboot. Everything looks ok.

The sensors are disconnected because, for example, the path /sys/devices/w1_bus_master1/28-022010ed82ae of the sensor on the 1Wire bus is no longer available. So currently no problem with the "VenusOS-TemperatureService" driver. We now have to see what the 1Wire service itself  is doing or why it is taking the sensors off the bus.

Can you check via SSH when the sensors are gone again if the appropriate folder of the sensor also gone in the folder /sys/devices/w1_bus_master1/28-XXXX ? 
If yes we have a 1Wire Problem not a "VenusOS-TemperatureService" problem. My service only intercepts the 1Wire data and brings the sensor data into the Venus OS dbus service and into the user interface.
I am looking to see if we can get a logfile from the 1Wire service to find out more.

Sarah-1331, 2024-01-21:
i agree i dont think its the driver as one of them never disconnects but i just cannot narrow it down and honestly am quite good with electronics

Sarah-1331, 2024-01-21:
4 just disconnected ```
2024-01-21 09:57:40.290489500 INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
2024-01-21 09:57:40.291074500 INFO:root:Loglevel set to INFO
2024-01-21 09:57:40.433965500 Traceback (most recent call last):
2024-01-21 09:57:40.433973500   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 403, in <module>
2024-01-21 09:57:40.433978500     initSettings(newSettings)
2024-01-21 09:57:40.433980500   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 263, in initSettings
2024-01-21 09:57:40.433985500     settings = SettingsDevice(
2024-01-21 09:57:40.433988500   File "/opt/victronenergy/dbus-modem/settingsdevice.py", line 51, in __init__
2024-01-21 09:57:40.433992500     raise Exception("The settings service com.victronenergy.settings does not exist!")
2024-01-21 09:57:40.434105500 Exception: The settings service com.victronenergy.settings does not exist!
2024-01-21 09:57:44.622391500 INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
2024-01-21 09:57:44.622736500 INFO:root:Loglevel set to INFO
2024-01-21 09:57:44.815598500 INFO:root:1Wire Sensor found with no Service -> Create:
2024-01-21 09:57:44.915107500 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.Wire_id01
2024-01-21 09:57:44.919001500 INFO:root:Add setting /Settings/Temperature/3c01d60728a6/TemperatureType [0, 0, 3]
2024-01-21 09:57:44.919892500 INFO:root:Add setting /Settings/Temperature/3c01d60728a6/CustomName ['', 0, 0]
2024-01-21 09:57:45.424907500 INFO:root:Retreived setting /Settings/Temperature/3c01d60728a6/TemperatureType /TemperatureType 0
2024-01-21 09:57:45.424917500 INFO:root:Retreived setting /Settings/Temperature/3c01d60728a6/CustomName /CustomName Fridge
2024-01-21 09:57:45.424923500 INFO:root:Created Service 1Wire ID: 1 Settings ID:1
2024-01-21 09:57:46.287942500 INFO:root:1Wire Sensor found with no Service -> Create:
2024-01-21 09:57:46.353083500 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.Wire_id02
2024-01-21 09:57:46.353092500 INFO:root:Add setting /Settings/Temperature/022010ed82ae/TemperatureType [0, 0, 3]
2024-01-21 09:57:46.353097500 INFO:root:Add setting /Settings/Temperature/022010ed82ae/CustomName ['', 0, 0]
2024-01-21 09:57:46.793719500 INFO:root:Retreived setting /Settings/Temperature/3c01d60728a6/TemperatureType /TemperatureType 0
2024-01-21 09:57:46.793728500 INFO:root:Retreived setting /Settings/Temperature/3c01d60728a6/CustomName /CustomName Fridge
2024-01-21 09:57:46.793734500 INFO:root:Retreived setting /Settings/Temperature/022010ed82ae/TemperatureType /TemperatureType 2
2024-01-21 09:57:46.793739500 INFO:root:Retreived setting /Settings/Temperature/022010ed82ae/CustomName /CustomName Externa ORIGl
2024-01-21 09:57:46.793745500 INFO:root:Created Service 1Wire ID: 2 Settings ID:2
2024-01-21 09:57:47.654085500 INFO:root:1Wire Sensor found with no Service -> Create:
2024-01-21 09:57:47.687517500 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.Wire_id03
2024-01-21 09:57:47.690929500 INFO:root:Add setting /Settings/Temperature/3c01d6076b91/TemperatureType [0, 0, 3]
2024-01-21 09:57:47.691675500 INFO:root:Add setting /Settings/Temperature/3c01d6076b91/CustomName ['', 0, 0]
2024-01-21 09:57:48.052415500 INFO:root:Retreived setting /Settings/Temperature/3c01d60728a6/TemperatureType /TemperatureType 0
2024-01-21 09:57:48.052427500 INFO:root:Retreived setting /Settings/Temperature/3c01d60728a6/CustomName /CustomName Fridge
2024-01-21 09:57:48.052948500 INFO:root:Retreived setting /Settings/Temperature/022010ed82ae/TemperatureType /TemperatureType 2
2024-01-21 09:57:48.053579500 INFO:root:Retreived setting /Settings/Temperature/022010ed82ae/CustomName /CustomName Externa ORIGl
2024-01-21 09:57:48.054588500 INFO:root:Retreived setting /Settings/Temperature/3c01d6076b91/TemperatureType /TemperatureType 2
2024-01-21 09:57:48.056179500 INFO:root:Retreived setting /Settings/Temperature/3c01d6076b91/CustomName /CustomName Internal
2024-01-21 09:57:48.057355500 INFO:root:Created Service 1Wire ID: 3 Settings ID:3
2024-01-21 09:57:48.943559500 INFO:root:1Wire Sensor found with no Service -> Create:
2024-01-21 09:57:48.980536500 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.Wire_id04
2024-01-21 09:57:48.984001500 INFO:root:Add setting /Settings/Temperature/3ce1e38061b1/TemperatureType [0, 0, 3]
2024-01-21 09:57:48.984639500 INFO:root:Add setting /Settings/Temperature/3ce1e38061b1/CustomName ['', 0, 0]
2024-01-21 09:57:49.403563500 INFO:root:Retreived setting /Settings/Temperature/3c01d60728a6/TemperatureType /TemperatureType 0
2024-01-21 09:57:49.404312500 INFO:root:Retreived setting /Settings/Temperature/3c01d60728a6/CustomName /CustomName Fridge
2024-01-21 09:57:49.405313500 INFO:root:Retreived setting /Settings/Temperature/022010ed82ae/TemperatureType /TemperatureType 2
2024-01-21 09:57:49.405690500 INFO:root:Retreived setting /Settings/Temperature/022010ed82ae/CustomName /CustomName Externa ORIGl
2024-01-21 09:57:49.406379500 INFO:root:Retreived setting /Settings/Temperature/3c01d6076b91/TemperatureType /TemperatureType 2
2024-01-21 09:57:49.407023500 INFO:root:Retreived setting /Settings/Temperature/3c01d6076b91/CustomName /CustomName Internal
2024-01-21 09:57:49.407742500 INFO:root:Retreived setting /Settings/Temperature/3ce1e38061b1/TemperatureType /TemperatureType 2
[... cut ...]
2024-01-21 09:57:54.396445500 INFO:root:Connected to dbus, and switching over to GLib.MainLoop() (= event based)
2024-01-21 10:21:28.296623500 INFO:root:W1-temp:28-3c01d60728a6 temperature interface disconnected
2024-01-21 10:21:28.314037500 INFO:root:W1-temp:28-3c01d6076b91 temperature interface disconnected
2024-01-21 10:21:28.314045500 INFO:root:W1-temp:28-3ce1e38061b1 temperature interface disconnected
2024-01-21 10:21:28.314049500 INFO:root:W1-temp:28-3ce104577d8f temperature interface disconnected
```

Rikkert-RS, 2024-01-21:
> i agree i dont think its the driver as one of them never disconnects but i just cannot narrow it down and honestly am quite good with electronics

Yes, I would also like my service to run well, and I would also like to know why it is not working so that I can set up a small wiki for the FAQ and small tips in the future.

Sarah-1331, 2024-01-21:
just edited the above my friend the folders have gone also

Rikkert-RS, 2024-01-21:
OK can you run the command: "dmesg" on ssh

Sarah-1331, 2024-01-21:
yes i can 

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.110-rpi-venus-4 (oe-user@oe-host) (arm-ve-linux-gnueabi-gcc (GCC) 9.5.0, GNU ld (GNU Binutils) 2.34.0.20200910) #1 SMP Wed Dec 20 00:28:59 UTC 2023
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.4
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2133 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s51916 r8192 d21812 u81920
[    0.000000] pcpu-alloc: s51916 r8192 d21812 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 877720K/970752K available (10240K kernel code, 1333K rwdata, 3100K rodata, 1024K init, 885K bss, 27496K reserved, 65536K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33798 entries in 100 pages
[    0.000000] ftrace: allocated 100 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3b4/0x588 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000024] Switching to timer-based delay loop, resolution 52ns
[    0.000326] Console: colour dummy device 80x30
[    0.000391] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000425] pid_max: default: 32768 minimum: 301
[    0.000627] LSM: Security Framework initializing
[    0.000870] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000899] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002293] cgroup: Disabling memory control group subsystem
[    0.002566] CPU: Testing write buffer coherency: ok
[    0.003056] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004296] Setting up static identity map for 0x100000 - 0x10003c
[    0.004491] rcu: Hierarchical SRCU implementation.
[    0.005605] smp: Bringing up secondary CPUs ...
[    0.006755] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008035] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009264] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009430] smp: Brought up 1 node, 4 CPUs
[    0.009472] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.009489] CPU: All CPU(s) started in HYP mode.
[    0.009505] CPU: Virtualization extensions available.
[    0.010587] devtmpfs: initialized
[    0.028677] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.028939] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.028974] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[... 285 lines cut ...]
[    7.930454] w1_master_driver w1_bus_master1: Attaching one wire slave 28.53754c1f64ff crc 1e
[    7.963767] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.972771] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    7.993371] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Nov  1 2021 00:37:25 version 7.45.241 (1a2f2fa CY) FWID 01-703fd60
[    8.140863] rtc-ds1307: probe of 1-0068 failed with error -121
[    8.447790] Bluetooth: Core ver 2.22
[... cut ...]
[    8.666829] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[   10.632809] watchdog: watchdog0: watchdog did not stop!
[   11.765397] NET: Registered protocol family 38
[   13.143305] device eth0 entered promiscuous mode
[   13.443793] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[   18.287178] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   20.310290] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[   33.755754] cam-dummy-reg: disabling
root@raspberrypi2:~#
```

Rikkert-RS, 2024-01-21:
OK the log is directly after the restart. As you can see, the 1Wire sensors are attached. You have to do this again if the sensors lose the connection again.

After  7.481933 seconds: -> w1_master_driver w1_bus_master1: Attaching one wire slave 28.3ce1e38061b1 crc a4

I just saw that you are using GPIO 18 for the 1Wire. Maybe there is interference with other drivers on the pin. Can you connect it to pin 4 or 26 for testing purposes?

Sarah-1331, 2024-01-21:
Sure can will do gpio 4 now

Rikkert-RS, 2024-01-21:
So I have also started a test.
I connected another sensor with a 50m Cat7 cable. 

Sensor Power on 5V
Pullup from 3.3V to data pin
GND

Is recognized immediately and is currently stable.
Here are a few pictures.

[image]
[image]

Sarah-1331, 2024-01-21:
Tested on gpio4 and disconnect again I have three remaining 4 folders in the devices have been removed also 
Again all sensors are connected direct to pi weirdly all the 1m ones have gone and the three 5m remain

Sarah-1331, 2024-01-21:
i tend to agree but all sensors are connect directly at the pi 

the 3 5m ones are still active and the 5 1m have disconnected the only way i could test further is cut the 5s down to 1m 

i feel if the 5m dont disconnect for 48hr ill just use theses 3 sensors

Rikkert-RS, 2024-01-22:
Here is also some info about fake Sensors
https://github.com/cpetrich/counterfeit_DS18B20

Sarah-1331, 2024-01-23:
https://www.ebay.co.uk/itm/124433541502?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=sf_h7LceQnu&sssrc=4429486&ssuid=45bWpJJDSEe&var=&widget_ver=artemis&media=COPY 
5m haven’t disconnected but the same listing 1m has

Sarah-1331, 2024-01-25:
Confirm 3days with only the 5m sensors and all good so far it must be the cheap crap in the 1m’s

morphias2004, 2024-01-25:
Cool. Maybe they use larger CSA or better quality wire in the longer ones?

Anything is possible with Chinese manufacturing.

I'll get some 5m ones to try and will cut one down to 1m to see if it continues to work as expected.

morphias2004, 2024-01-25:
FWIW, I'm using pin 27.

Sarah-1331, 2024-01-25:
I cut one down to a meter and it never failed in any test  I found changing gpio had no effect either

morphias2004, 2024-01-25:
```
root@raspberrypi2:/u-boot# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.110-rpi-venus-4 (oe-user@oe-host) (arm-ve-linux-gnueabi-gcc (GCC) 9.5.0, GNU ld (GNU Binutils) 2.34.0.20200910) #1 SMP Tue Dec 5 23:19:16 UTC 2023
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2133 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s51916 r8192 d21812 u81920
[    0.000000] pcpu-alloc: s51916 r8192 d21812 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 root=/dev/mmcblk0p3 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 877720K/970752K available (10240K kernel code, 1333K rwdata, 3100K rodata, 1024K init, 885K bss, 27496K reserved, 65536K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33798 entries in 100 pages
[    0.000000] ftrace: allocated 100 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3b4/0x588 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000024] Switching to timer-based delay loop, resolution 52ns
[    0.000325] Console: colour dummy device 80x30
[    0.000390] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000425] pid_max: default: 32768 minimum: 301
[    0.000629] LSM: Security Framework initializing
[    0.000874] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000902] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002297] cgroup: Disabling memory control group subsystem
[    0.002571] CPU: Testing write buffer coherency: ok
[    0.003067] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004305] Setting up static identity map for 0x100000 - 0x10003c
[    0.004495] rcu: Hierarchical SRCU implementation.
[    0.005619] smp: Bringing up secondary CPUs ...
[    0.006766] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008048] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009273] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009440] smp: Brought up 1 node, 4 CPUs
[    0.009482] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.009499] CPU: All CPU(s) started in HYP mode.
[    0.009515] CPU: Virtualization extensions available.
[    0.010600] devtmpfs: initialized
[    0.028782] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.029041] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.029078] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.032085] pinctrl core: initialized pinctrl subsystem
[... 319 lines cut ...]
[    8.729121] i2c /dev entries driver
[    8.774950] EXT4-fs (mmcblk0p3): re-mounted. Opts: (null)
[   10.729911] watchdog: watchdog0: watchdog did not stop!
[   11.808092] NET: Registered protocol family 38
[   13.081288] device eth0 entered promiscuous mode
[   13.081425] lan78xx 1-1.1.1:1.0 eth0: kevent 4 may have been dropped
[   13.103255] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   13.110455] IPv6: ADDRCONF(NETDEV_CHANGE): ll-eth0: link becomes ready
[... cut ...]
[   17.937459] rtc-ds1307 1-0068: registered as rtc0
[   17.943304] rtc-ds1307 1-0068: setting system clock to 2024-01-04T08:22:58 UTC (1704356578)
[   18.220300] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   20.670322] can: controller area network core
[   20.686227] NET: Registered protocol family 29
[   20.795157] IPv6: ADDRCONF(NETDEV_CHANGE): can10: link becomes ready
[   22.461536] can: raw protocol
[   23.801634] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   33.755011] cam-dummy-reg: disabling

```

morphias2004, 2024-01-25:
Directly following a shutdown, power off and power on:

```
Last login: Thu Jan 25 22:48:54 2024
root@raspberrypi2:~# ls /sys/bus/w1/devices/
28-3c01f0952677  w1_bus_master1
root@raspberrypi2:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.110-rpi-venus-4 (oe-user@oe-host) (arm-ve-linux-gnueabi-gcc (GCC) 9.5.0, GNU ld (GNU Binutils) 2.34.0.20200910) #1 SMP Tue Dec 5 23:19:16 UTC 2023
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
[    0.000000] random: fast init done
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x37400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   Normal   empty
[    0.000000]   HighMem  empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000] On node 0 totalpages: 242688
[    0.000000]   DMA zone: 2133 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s51916 r8192 d21812 u81920
[    0.000000] pcpu-alloc: s51916 r8192 d21812 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 240555
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 root=/dev/mmcblk0p3 rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 877720K/970752K available (10240K kernel code, 1333K rwdata, 3100K rodata, 1024K init, 885K bss, 27496K reserved, 65536K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 33798 entries in 100 pages
[    0.000000] ftrace: allocated 100 pages with 3 groups
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Rude variant of Tasks RCU enabled.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x3b4/0x588 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000024] Switching to timer-based delay loop, resolution 52ns
[    0.000326] Console: colour dummy device 80x30
[    0.000390] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000425] pid_max: default: 32768 minimum: 301
[    0.000633] LSM: Security Framework initializing
[    0.000875] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.000903] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002304] cgroup: Disabling memory control group subsystem
[    0.002577] CPU: Testing write buffer coherency: ok
[    0.003071] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004314] Setting up static identity map for 0x100000 - 0x10003c
[    0.004504] rcu: Hierarchical SRCU implementation.
[    0.005630] smp: Bringing up secondary CPUs ...
[    0.006780] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.008058] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.009287] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009454] smp: Brought up 1 node, 4 CPUs
[    0.009496] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.009513] CPU: All CPU(s) started in HYP mode.
[    0.009529] CPU: Virtualization extensions available.
[... 319 lines cut ...]
[    8.003908] gs_usb 1-1.2.3:1.0 can10: renamed from can0
[    8.287764] F2FS-fs (sda): Magic Mismatch, valid(0xf2f52010) - read(0x0)
[    8.294488] F2FS-fs (sda): Can't find valid F2FS filesystem in 1th superblock
[    8.302572] F2FS-fs (sda): Magic Mismatch, valid(0xf2f52010) - read(0x0)
[    8.309317] F2FS-fs (sda): Can't find valid F2FS filesystem in 2th superblock
[    8.632691] FAT-fs (sda1): Volume was not properly unmounted. Some data may be corrupt. Please run fsck.
[    8.759493] i2c /dev entries driver
[    8.805025] EXT4-fs (mmcblk0p3): re-mounted. Opts: (null)
[   10.827447] watchdog: watchdog0: watchdog did not stop!
[   11.889523] NET: Registered protocol family 38
[... cut ...]
[   18.848252] i2c i2c-1: new_device: Instantiated device ds1307 at 0x68
[   18.863854] rtc-ds1307 1-0068: registered as rtc0
[   18.870008] rtc-ds1307 1-0068: setting system clock to 2024-01-25T22:48:54 UTC (1706222934)
[   19.050784] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
[   20.714130] can: controller area network core
[   20.718812] NET: Registered protocol family 29
[   20.779720] IPv6: ADDRCONF(NETDEV_CHANGE): can10: link becomes ready
[   21.531513] can: raw protocol
[   33.754856] cam-dummy-reg: disabling

```

morphias2004, 2024-01-25:
> Here is also some info about fake Sensors https://github.com/cpetrich/counterfeit_DS18B20

Based on that, I believe I have a clone. :(

```
root@raspberrypi2:~# ls /sys/bus/w1/devices/
28-3c01f0952677  w1_bus_master1
```

MM0RBZ, 2024-02-13:
Thanks Rikkert.
I've added a 6.8uF capacitor across the supply lines and it's cleaned up the ripple, I'll leave it for 24hrs and see if it's helped but one sensor was working fine with the ripple there so I won't hold my breath.
I note that you use 50m of cat7 between the pi and the sensors, I have a gut feeling that having a longer cable there might help .
Theory says it should be short but maybes long might be better in practice.

MM0RBZ, 2024-02-14:
better ripple.

[image]

MM0RBZ, 2024-02-15:
I have ordered 3 of these... https://thepihut.com/products/ds18b20-one-wire-digital-temperature-sensor
they claim to be genuine Maxim chips so I'll try them and report back.

Sarah-1331, 2024-02-15:
Quick update from me mine haven’t disconnected once since using the 5m ones

MM0RBZ, 2024-02-15:
Thanks Sarah. :-)
I think you maybe got lucky and a good batch of sensors was used to make your 5m ones.
Would you mind posting their serial numbers please?

Sarah-1331, 2024-02-15:
Am unsure the SN but think it would be one of these numbers ???

TemperatureType /TemperatureType 1
@4000000065b843c429dc25fc INFO:root:Retreived setting /Settings/Temperature/022010ed82ae/CustomName /CustomName Fridge
@4000000065b843c429e5cabc INFO:root:Retreived setting /Settings/Temperature/485e4c1f64ff/TemperatureType /TemperatureType 2
@4000000065b843c429edecc4 INFO:root:Retreived setting /Settings/Temperature/485e4c1f64ff/CustomName /CustomName External
@4000000065b843c429f6e5a4 INFO:root:Retreived setting /Settings/Temperature/53754c1f64ff/TemperatureType /TemperatureType 2
@4000000065b843c42a0c658c INFO:root:Retreived setting /Settings/Temperature/53754c1f64ff/CustomName /CustomName Internal

MM0RBZ, 2024-02-16:
Thanks Sarah.
it looks like your serial numbers are 28-022010ed82ae,  28-485e4c1f64ff and 28-53754c1f64ff.

MM0RBZ, 2024-02-19:
Update-
I got some genuine Maxim sensors from "ThePiHut" and they have solved the problem. :-)

espenfj, 2024-07-09:
Hi!

I am experimenting with some temp-sensors in my caravan. I tried to be clever and created an expandable/module based build using TP-cables and RJ45-connectors. The idea is to use wall-boxes or joint pieces for the connections, and custom lenght TP-cable. 

If my professional drawing is understandable, you'll get the idea:
[image]

(1-wire = DS18B20).
My problem right now is that one of the two DS18B20's in the first box, keeps failing after a short period of time. 30-60 minutes. Venus is reporting "open circuit" on the "disconnected" sensor.
The other two seems to be ok. 

My question for you is;
- May the reason be that I use one shared resistor for the two sensors?
      - Can I put one single resistor by the very first RJ45-connector and drop the rest of them?
- is it a bad sensor? 
- or should I redo my soldering? (my soldering skills are about the same level as my drawing skills).

Sarah-1331, 2024-07-09:
In my experience it’s cheap sensors buddy

Rikkert-RS, 2024-07-09:
Hello

In the 1Wire Bus, you should only use one pullup resistance on the data line

Best regards

espenfj, 2024-07-10:
Thank you for quick answers, guys!
I split my 2xtemp-device into two singles and put a pullup resistance on the connector nearest the RPi. It really made no difference, but I am now able to isolate the sensor that keeps failing. I guess I'll have to buy a new sensor and try again after the upcoming trip :)

Rikkert-RS, 2025-01-15:
I'll close this topic here:

To summarize, in most cases it's due to the following problems:

- bad connection / connectors
- wrong cable
- bad sensors (fake)
- wrong pull-up resistors configuration

closed 2025-01-15 by Rikkert-RS