Rikkert-RS/VenusOS-TemperatureService #12: Negetive temperature not reporting

Julesmr, 2023-11-17:
I hope you can shed some light on this. I have several DS18B20 sensors in my van on a Venus OS RPI. Whichever one I put in the freezer disconnects once its below 0 degrees C. Are you aware of such an issue. Thanks for your time.

GustavoGustik, 2023-11-19:
I have the same issue. My DS18B20 has temperature range -55 to 125 °C and it shows this, when I put sensor to the freezer:
[image]

Julesmr, 2023-11-19:
Yes that's just the same as mine. I was looking at a posting yesterday day within a RasPI forum and they found a code error - It wasn't Venus OS though. I am going to check the code one evening and see if it's a similar issue. Or, it's a fake sensor that has limited range.

I will post any findings here.

AdventureHaus, 2023-11-29:
Any chance you resolved this issue. With the sudden drop in temperature here in the uk ive noticed omy one wire sensors read ok when the temp gets to 0

Many thanks

Rikkert-RS, 2023-11-29:
Hello everyone,

I think that with version 0.51 I have solved the problem with the negative temperatures not being displayed. Please update to v0.51.
Attention, if you use a pin other than Pin26 then adjust it in u-boot/config.txt after installation.

Release v0.51 -> https://github.com/Rikkert-RS/VenusOS-TemperatureService/releases/tag/v0.51

best regards Rikkert-RS

AdventureHaus, 2023-11-30:
Works perfectly thankyou 🙏 
I use pin 23, I didnt have to edit the configuration.txt file after the update. The sensors read negative after reboot.

Again thankyou for your efforts on this 

[image]

Julesmr, 2023-11-30:
Looking forward to trying this. Thank you for your work.

Julesmr, 2023-11-30:
Works perfectly 👍

closed 2023-12-10 by Rikkert-RS