Rikkert-RS/VenusOS-TemperatureService #13: help with pi temp flashing

sarah1331sarah, 2023-12-08:
i have installed latest version 
Thank you for your addition to the community 

i have not installed anything except guimodes and setup helper 

raspberry pi 3 keeps flashing up its temp then dissconnecting (not connected) then flashing up again maybe every 500ms 

have i done someting wrong i am happt for it to be there (would pefer it not to be ) but it disconnecting etc would love to stop

Rikkert-RS, 2023-12-10:
Hello,

this is a strange behavior.
Do you have connected any temperature sensors to pin 26?
Is Pin26 free or is there something else connected there?

Can you view the log file via SSH:
/data/log/VenusOS-TemperatureService/current

sarah1331sarah, 2023-12-10:
pin 26 is free at the moment sensors due for delivery tomorow

Rikkert-RS, 2023-12-10:
Is it still flashing?

Can you read out the log file for me?

Rikkert-RS, 2023-12-10:
Apparently the service crashes at some point, but very difficult to check without a log file

sarah1331sarah, 2023-12-10:
I will do thanks for helping I’ll update tomorrow

sarah1331sarah, 2023-12-12:
hi thanks for the help 

so i have connected temp sensor to gpio26 
and it doesnt show also the pi cpu is still flashing then dissconnecting 

here is the log from /data/log/tempservice/current
```
@40000000657871ea042fc124     fd = open('/sys/devices/w1_bus_master1/w1_master_slaves','r')
@40000000657871ea05b53074 FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/w1_bus_master1/w1_master_slaves'
@40000000657871eb1e6889cc INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
@40000000657871eb1e70907c INFO:root:Loglevel set to INFO
@40000000657871eb1fb6ef44 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.RPi_cpu_id01
@40000000657871eb1fb70e84 INFO:root:Add setting /Settings/Temperature/1/TemperatureType [0, 0, 3]
@40000000657871eb1fb7220c INFO:root:Add setting /Settings/Temperature/1/CustomName ['', 0, 0]
@40000000657871eb2d779084 INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 0
@40000000657871eb2d7faabc INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName 
@40000000657871eb2d897a74 INFO:root:cpu temperature interface connected
@40000000657871eb2e426264 Traceback (most recent call last):
@40000000657871eb2e4279d4   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 408, in <module>
@40000000657871eb2e428d5c     update()
@40000000657871eb2e429144   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 59, in update
@40000000657871eb2e42a4cc     update_W1()
@40000000657871eb2e42a8b4   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 155, in update_W1
@40000000657871eb2e42bc3c     fd = open('/sys/devices/w1_bus_master1/w1_master_slaves','r')
@40000000657871eb2e445e34 FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/w1_bus_master1/w1_master_slaves'
@40000000657871ec2fe5d54c INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
@40000000657871ec2fecbed4 INFO:root:Loglevel set to INFO
@40000000657871ec3084f794 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.RPi_cpu_id01
@40000000657871ec30b79884 INFO:root:Add setting /Settings/Temperature/1/TemperatureType [0, 0, 3]
@40000000657871ec30c154b4 INFO:root:Add setting /Settings/Temperature/1/CustomName ['', 0, 0]
@40000000657871ec348acc4c INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 0
@40000000657871ec349280f4 INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName 
@40000000657871ec349c0e44 INFO:root:cpu temperature interface connected
@40000000657871ec3548adac Traceback (most recent call last):
@40000000657871ec3548d0d4   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 408, in <module>
@40000000657871ec3548e074     update()
@40000000657871ec3548e844   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 59, in update
@40000000657871ec3549039c     update_W1()
@40000000657871ec35490b6c   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 155, in update_W1
@40000000657871ec35491ef4     fd = open('/sys/devices/w1_bus_master1/w1_master_slaves','r')
@40000000657871ec354aa97c FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/w1_bus_master1/w1_master_slaves'
@40000000657871ed342a1a64 INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
@40000000657871ed34312714 INFO:root:Loglevel set to INFO
@40000000657871ed34f3c364 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.RPi_cpu_id01
@40000000657871ed352b7d04 INFO:root:Add setting /Settings/Temperature/1/TemperatureType [0, 0, 3]
@40000000657871ed35350a54 INFO:root:Add setting /Settings/Temperature/1/CustomName ['', 0, 0]
@40000000657871ed37f2ab84 INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 0
@40000000657871ed37fab234 INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName 
@40000000657871ed38042bfc INFO:root:cpu temperature interface connected
@40000000657871ed3892df3c Traceback (most recent call last):
@40000000657871ed3892f6ac   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 408, in <module>
@40000000657871ed38930a34     update()
@40000000657871ed38931204   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 59, in update
@40000000657871ed389321a4     update_W1()
@40000000657871ed38932974   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 155, in update_W1
@40000000657871ed38933914     fd = open('/sys/devices/w1_bus_master1/w1_master_slaves','r')
@40000000657871ed3895b9b4 FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/w1_bus_master1/w1_master_slaves'
@40000000657871ee38dc732c INFO:root:/data/VenusOS-TemperatureService/dbus-i2c.py is starting up - python 3
@40000000657871ee38e3974c INFO:root:Loglevel set to INFO
@40000000657871ee39edf6b4 INFO:root:registered ourselves on D-Bus as com.victronenergy.temperature.RPi_cpu_id01
@40000000657871ee3a1eb344 INFO:root:Add setting /Settings/Temperature/1/TemperatureType [0, 0, 3]
@40000000657871ee3a26d934 INFO:root:Add setting /Settings/Temperature/1/CustomName ['', 0, 0]
@40000000657871ef01647fac INFO:root:Retreived setting /Settings/Temperature/1/TemperatureType /TemperatureType 0
@40000000657871ef016c3454 INFO:root:Retreived setting /Settings/Temperature/1/CustomName /CustomName 
@40000000657871ef01759a94 INFO:root:cpu temperature interface connected
[... cut ...]
@40000000657871f926022404     update()
@40000000657871f926022fbc   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 59, in update
@40000000657871f926023f5c     update_W1()
@40000000657871f92602472c   File "/data/VenusOS-TemperatureService/dbus-i2c.py", line 155, in update_W1
@40000000657871f9260256cc     fd = open('/sys/devices/w1_bus_master1/w1_master_slaves','r')
@40000000657871f926042f74 FileNotFoundError: [Errno 2] No such file or directory: '/sys/devices/w1_bus_master1/w1_master_slaves
```

Rikkert-RS, 2023-12-12:
Hello sarah1331sarah,

OK, that helps, I can already see the problem.
First of all, it is because there is no temp sensor connected and some 1Wire Folder does not exist.

But I can't look at it until tomorrow.

But you can just write to me how you connected the sensors to Pin26. Because with sensor the error shouldn't occur

best regards

sarah1331sarah, 2023-12-12:
data wire direct to pin 37 GPIO26 5v to red gnd to black

Rikkert-RS, 2023-12-12:
no pullup?

you need a 4,7kOhm pullup from 5V to Data Pin

sarah1331sarah, 2023-12-12:
will try that not sorry for my stupidity

sarah1331sarah, 2023-12-12:
still nothing shown i think its the file and folder ?

Rikkert-RS, 2023-12-12:
https://github.com/Rikkert-RS/VenusOS-TemperatureService/issues/7#issuecomment-1596296044

There are some pictures with 1Wire connection

But there is still an error if no sensor is connected and only the CPU temp is used. I will fix the error tomorrow

sarah1331sarah, 2023-12-12:
i have the 4.7k now i was going to ask if i add more do i need to adjust the value but #7 shows i dont thanks again 

the cpu temp shows then "disconnects" then shows etc 

if i knew the file its requesting i can make the folder and add it ?

can i take the time here to show my apreacation for you help

Rikkert-RS, 2023-12-12:
> still nothing shown i think its the file and folder ?

Which type of sensor you using?

sarah1331sarah, 2023-12-12:
DS18B20

sarah1331sarah, 2023-12-12:
i have just started again with my test pi3 and its now working hopefully its notsomethig else i install from the setuphelper conflicting

sarah1331sarah, 2023-12-12:
no folder w1_bus_master1 but its just gone again after installing items from setup helper i will install 1 by 1 and find the conflict

Rikkert-RS, 2023-12-12:
Which version of Temp Service is installed in your setup helper?

Rikkert-RS, 2023-12-12:
OK, that sounds like a GPIO conflict.

I have to check it on the PC

I'll let you know when I know more

sarah1331sarah, 2023-12-12:
100% is if i edit setup to make it run on gpio4 it works again with GPIOsetup 
although i need 4 for rtc lol ill get there am sure

sarah1331sarah, 2023-12-12:
thanks again changed setup 
```
        echo "#### Change 1-Wire Temperatursensor on Pin20"  >> "$configFile"
		echo "dtoverlay=w1-gpio,gpiopin=20" >> "$configFile"
		echo "#### end change 1-Wire Pin20" >> "$configFile"
        filesUpdated=true
    fi
	
fi

#### uninstalling - check scriptAction again
# if an install step failed package needs to be removed
if [ $scriptAction == 'UNINSTALL' ] ; then
    logMessage "++ Uninstalling VenusOS Temperature Service"

    removeService $packageName
	
	restoreActiveFile "$overlayFile"
	
	# remove mods from configFile - do not use restore in case other mods were made manually
    if [ -f "$configFile" ]; then
        if [ $(grep -c "#### Change 1-Wire Temperatursensor on Pin20" "$configFile") != 0 ]; then
            sed -i -e '/#### Change 1-Wire Temperatursensor on Pin20/,/#### end change 1-Wire Pin20/d' "$configFile"
```

no conflicts at all 

thank you again

Rikkert-RS, 2023-12-12:
I'm back on the PC now, but I can no longer understand your current configuration exactly.

1. Which software packages do you still have installed next to TemperatureService with or without Setup Helper? 
2. How is the sensor connected now?
3. Do I understand correctly that it is now working for you?

I have solved the problem with crashing service on my Testsetup with no sensors connected.
But I will only create a new version if the discussion indicates that no further changes are necessary

Unfortunately, there are always conflicts with the GPIOs, which is unfortunately normal and difficult to solve.

!!! Important !!!
if you want to change the GPIO you don't need to edit the Setup File and reinstall. 
(This setting also survives an TemperatureService update but not an Venus OS update)

Goto /u-boot/config.txt

and Change only the Pin Number . Do not change the text that begins with ####

```
#### Change 1-Wire Temperatursensor on Pin26
dtoverlay=w1-gpio,gpiopin=26 <----- just make changes here
#### end change 1-Wire Pin26

```

sarah1331sarah, 2023-12-12:
Ah so I have everything working I ended up changing the setup I didn’t know I could change the config.txt thank you now I know for future 

settled on pin 20 as gpio setup doesn’t use this pin either 
So after a firmware update just reinstall using setup helper and edit the config 

perfect

Rikkert-RS, 2023-12-12:
OK fine

Now i Understand that you use rpigpiosetup from Setup Helper

Here is another soleved issue with the same topic. I'll probably add this to the readme as a note.
https://github.com/Rikkert-RS/VenusOS-TemperatureService/issues/5#issuecomment-1556277827

I will immediately create version v0.52 with the fix.
If you change the setup file it will survive a Venus OS update a change in uboot/config.txt not but in principle, both are possible
In the future I plan to bring the port setup into the Gui

nice that it works now.

Rikkert-RS, 2023-12-12:
@sarah1331sarah if you want you can try v0.52. It should now run with and without sensors.
It is on Setup Helper ready for install / update. After Update check your GPIO in Config.txt is still 20.

sarah1331sarah, 2023-12-13:
all working perfect thanks again

closed 2023-12-13 by sarah1331sarah