Rikkert-RS/VenusOS-TemperatureService #16: [NOT A ISSUE] Is your module compatible with RpiGpioSetup?

mvincm, 2024-01-26:
Hello,

Thanks for your work and great module! I wonder if your module is compatible with RpiGpioSetup created by kwindrem. I'm talking about this one: https://github.com/kwindrem/RpiGpioSetup.

I would like to use one (or two) DS18B20 and some GPIO as a relay control.

Thanks!
MvincM

Rikkert-RS, 2024-01-26:
Hello MvincM,

yes it is compatible
Unfortunately, there are always conflicts with the GPIOs and Addon's, which is unfortunately normal and difficult to solve.

There are two ways to solve this
1. change the GPIO Setup in RpiGpioSetup to release Pin26
    The default Pin (GPIO) after installation is 26 

2. change the Pin from 26 to XX
    but you have to modify the config.txt manually after installation .

I am working on a solution to change the corresponding 1Wire Pin from Venus OS settings menu, but it is not ready jet.

You have to logon with SSH to your System and change the pin settings in this file: "/u-boot/config.txt"

Only change the pin number . Do not change the text that begins with #### and around it
Restart and connect your sensors.

```
#### Change 1-Wire Temperatursensor on Pin26
dtoverlay=w1-gpio,gpiopin=26 <----- just make changes here
#### end change 1-Wire Pin26
```

best regards 

Rikkert-RS

Rikkert-RS, 2024-01-26:
As additional info:

you can also customize the pins in RpiGpioSetup created by kwindrem:

This is in this file here:
https://github.com/kwindrem/RpiGpioSetup/blob/main/FileSets/gpio_list

You can also do this on your system with SSH
You have comment line 45

[image]

And here you can see that kwindrem already has an extra GPIO file if a CAN HAT is used on the Raspberry. 

[image]

mvincm, 2024-01-26:
All I want to know! Thanks! I will do it! Once again thanks for the quick help!

closed 2024-01-26 by mvincm