Rikkert-RS/VenusOS-TemperatureService #21: Installation failed on Raspi 4 (VenusOS 3.52)

moe9584, 2025-01-04:
After updating Venus OD from v3.34 to the latest v3.52 during installation following errors are shown on Remote Console:

```VenusOS-TemperatureService install failed - unkonwn error 127
/data/VenusOS-TemperatureService/setup: line18: 
/data/SetupHelper/Common Ressources: No such file or directory
/data/VenusOS-TemperatureService/setup: line24: [: ==: unary operator expected
/data/VenusOS-TemperatureService/setup: line33: [: ==: unary operator expected
/data/VenusOS-TemperatureService/setup: line54: [: ==: unary operator expected
/data/VenusOS-TemperatureService/setup: line90: endScript: command not found

moe9584, 2025-01-04:
I made the changes suggest from the latest PULL REQUEST from lennardini and everything is working now!

closed 2025-01-04 by moe9584

Rikkert-RS, 2025-01-11:
Hello, great that the PR from lennardini has already helped you. I will check it in the next few days and create a new version. But I want to test this in my setup first