Rikkert-RS/VenusOS-TemperatureService #22: installation failed venus OS 3.53

Thierry43, 2025-01-14:
Hello,

It seems that from version 3.50 up of Venus OS the VenusOS-TemperatureService instalaltion is failling.
See details below:
[image]

Can you help me fix it ?
Thanks

r3dmax, 2025-01-15:
same on 3.52 ....and you can not uninstall/delete it ...once you proceed with remove, 5 seconds later it installs itself back. I wanted to check if i am doing something wrong before i saw this post that others are also having problems, by installing and uninstalling it , but i cant get it off anymore...i even proceeded with remove and unplugged the raspberry but as soon as it came up it installed itself (tried to reboot, reboot gui and restart package manager) ....well ...its there now....hope you can update it to make it work...just bought 3 sensors for this script and cant use them :(

Rikkert-RS, 2025-01-15:
Hello everyone,
Which version of Setup Helper have you installed?
I will test this on my system

Thierry43, 2025-01-15:
Hello, I am using the latest version of setup helper

r3dmax, 2025-01-15:
setup helper v8.24

Rikkert-RS, 2025-01-15:
@r3dmax and @Thierry43 

ok I have created a new version.

Use tag "latest" or "v0.53"

best regards Rikkert-RS

r3dmax, 2025-01-16:
great, ..works!!...you are a magician :) .... thnx

closed 2025-10-17 by Rikkert-RS