Rikkert-RS/VenusOS-TemperatureService #23: [BUG] Need to add properly register Device Instance

morphias2004, 2025-02-24:
The current means of allocating a Device Instance results in duplicate instance numbers when using Node Red virtual devices, etc.

It is due to VenusOS-TemperatureService not checking for and properly a the device instance.

Please see comments and explanation here:

https://github.com/victronenergy/node-red-contrib-victron/issues/216#issuecomment-2677668149

Rikkert-RS, 2025-02-25:
Hello

I will check this tomorrow

morphias2004, 2025-03-30:
> Hello
> 
> I will check this tomorrow

Hey mate,

Just wondering how you went?

morphias2004, 2025-03-31:
I have tried at length to implement `addSettings` but it's beyond my skill level.

As a temporary work-around, I have edited the following lines to stop the usage of the 100 range:

[image]

[image]

Rikkert-RS, 2025-10-17:
Hi again

Same here I haven't had much time to take care of the repo.
I hope I can continue with it now. I also need to update everything for testing purposes.

I need to check how to comply with the Victron ID allocation