Rikkert-RS/VenusOS-TemperatureService #24: Add support for multiple GPIO pins

morphias2004, 2025-03-30:
If you want to use multiple GPIO pins for 1-Wire sensors, only one pin is brought through the Venus OS.

I have just made my life easier on my RPi by adding this expansion board so that I can use Cat5/6 cables to plug in external devices:

https://czh-labs.com/products/rj45-gpio-pinout-breakout-board-for-raspberry-pi-1a-1b-2b-3b-3b-3a-4b-zero-w

[image]

I have a mixture of 1-Wire and I2C sensors.

All can be interrogated on the command line, but only one 1-Wire GPIO's data appears in Venus OS.

[image]

This is my /u-boot/config.txt

[image]

I'm no coder, but I see in the dbus-i2c.py file that it is hard coded for  `w1_bus_master1`.

Is it possible for the code to be updated to print in all 1-wire masters?

morphias2004, 2025-03-30:
I created a PR for this now.

https://github.com/Rikkert-RS/VenusOS-TemperatureService/pull/25

Rikkert-RS, 2025-10-17:
Hi

I haven't had much time to take care of the repo.
I hope I can continue with it now. I also need to update everything for testing purposes.
I'll take a look at your PR. Thanks for the support.

I'd also like to be able to turn the various functions on and off in the GUI, like CPU/1Wire/i2c, etc. But the new GUI in Venus OS is different now.

best regards Rikkert

bobmey64, 2026-08-31:
Latest (V0.53) did not work for my 2 bus configuration.  V0.55 fixed these issues and is working well.